smaffer/vgax #4: Will this work with ATmega2560?

snobu, 2016-03-01:
Is ATmega2560 supported?
I can't seem to make it work, triple checked the wiring, is there anything i need to change in the lib?
It does compile and upload just fine but the monitor doesn't detect any signal.
Thanks!

smaffer, 2016-03-01:
Hello Adrian,
i own an atmega2560 but i have not tested it yet.
In the future i will test it and add support to it, with a bigger vertical resolution.
If you find some modification to be made, let me know.

Thank you!
Sandro Maffiodo

snobu, 2016-03-01:
In theory if i connect to the PORTD pins on the Mega (which are different vs. Uno), i should be fine, right?
[image]

snobu, 2016-03-01:
On second thought there aren't enough PORTD pins exposed on the Mega, maybe i should try going for PORTA or PORTC.

Is this where the magic happens?
https://github.com/smaffer/vgax/blob/58a7343188db070f2afb6f0da8f4700ac866e7fc/VGAX.cpp#L155

smaffer, 2016-03-01:
Yes you need to use another port. Keep in mind that the library write the 2 most significant bits of an 8bit port. (bit6 and bit7). The others 4 bits will be written too with garbage.

snobu, 2016-03-02:
Your work on this library is outstanding and your understanding of the inner-workings of AVRs is seriously hard core. Congrats on both.

Now, back on topic, i've tried moving __HSYNC__ and __VSYNC__ to PIN 2 and 3 on the Mega to free up __PORTB__ for Red and Green.

[image]

Changed __COLORPIN0__ and __COLORPIN1__ to:
```
//These two pin cannot be modified without modify the HSYNC assembler code
#define COLORPIN0 12
#define COLORPIN1 13
```

And
https://github.com/smaffer/vgax/blob/58a7343188db070f2afb6f0da8f4700ac866e7fc/VGAX.cpp#L155 to __PORTB__:

```c
: [port] "I" (_SFR_IO_ADDR(PORTB)),
```

Still no go. Maybe i need to adjust something in the assembly code too?
Also, i'm not sure which pins MUST be PWM pins?

snobu, 2016-03-02:
Hmm, TIMERs are hardwired to pins, so i need to take care of that.

http://playground.arduino.cc/Main/TimerPWMCheatsheet
```
For Arduino Mega: (tested on Arduino Mega 2560)
timer 0 (controls pin 13, 4)
timer 1 (controls pin 12, 11)
timer 2 (controls pin 10, 9)
timer 3 (controls pin 5, 3, 2)
timer 4 (controls pin 8, 7, 6)
```

smaffer, 2016-03-02:
Hello!
Thank you for your compliments!! :)

The HSYNC and VSYNC cannot be moved. Are PWM signals and if you move them you need to change the begin() method. 
Instead of move HSYNC and VSYNC try to use another port for the color signals, PORTK or PORTC?

The AUDIO pin can be moved on any pin but need a modification inside the assembler that generate the audio signal.

smaffer, 2016-03-02:
Aaron! So on ATMEGA2560 you need to change HSYNC and VSYNC…

smaffer, 2016-03-02:
http://sphinx.mythic-beasts.com/~markt/ATmega-timers.html

snobu, 2016-03-02:
I now have the VSYNC and HSYNC part figured out (thanks!), i have sync, monitor says `640x480 31.2kHz 60Hz` but screen is all black. Tried PORTA, PORTC, PORTF to no effect (Won't compile with PORTK). Not sure what i'm missing now.

smaffer, 2016-03-02:
Well!
Have you tried PB7(D13) and PB6(D12) ?
PC7(D30) and PC6(D31) ?

snobu, 2016-03-02:
Yes. 

PORTB D13, D12 -> blank screen
PORTC D30, D31 -> blank screen
PORTA D28, D29 -> blank screen
PORTL D42, D43 -> blank screen

Not sure what to check next.

snobu, 2016-03-04:
Monitor connectivity is good. Everything works if i flash to a UNO.
Any idea what should i look into next? I feel like i reached a dead end here.

crazytiti, 2016-04-28:
Did you succeed in using VGAX on mega2560 ?
can you post all the modificatino that have to be done, maybe a pull request on the github :
https://github.com/smaffer/vgax

snobu, 2016-04-29:
I gave up midway through. In theory it should work. I've only got far enough to get Vsync, Hsync and a blank screen. I'm sure it's doable just needs more effort.

vrxfile, 2016-06-01:
Hello!
I have the same problem.
Tried all ports. The screen is blank. VSYNC and HSYNC signals work normally.

Tried change port on Arduino UNO to PORTC - works normally. But in MEGA doesn't work.

maujabur, 2016-09-14:
Hi! I managed to make it work on the mega, I just couldn't make the anti jitter routine work? could anybody help? I don't know why it doesn't work on the mega

this is the part that needs to be commented out: 
    //interrupt jitter fix (needed to keep signal stable)
    //code from https://github.com/cnlohr/avrcraft/tree/master/terminal
    //modified from 4 nop align to 8 nop align
    #define DEJITTER_OFFSET 1
    #define DEJITTER_SYNC -3
    asm volatile(
      "     lds r16, %[timer0]    \n\t" //
      //"   add r16, %[toffset]   \n\t" //
      "     subi r16, %[tsync]    \n\t" //
      "     andi r16, 7           \n\t" //
      "     call TL               \n\t" //
      "TL:                        \n\t" //
      "     pop r31               \n\t" //
      "     pop r30               \n\t" //
      "     adiw r30, (LW-TL-5)   \n\t" //
      "     add r30, r16          \n\t" //
      //"   adc r31, __zero_reg__ \n\t" //
      "     ijmp                  \n\t" //
      "LW:                        \n\t" //
      "     nop                   \n\t" //
      "     nop                   \n\t" //
      "     nop                   \n\t" //
      "     nop                   \n\t" //
      "     nop                   \n\t" //
      "     nop                   \n\t" //
      "     nop                   \n\t" //
      //"   nop                   \n\t" //
      "LBEND:                     \n\t" //
    :
    : [timer0] "i" (&TCNT0),
      [toffset] "i" ((uint8_t)DEJITTER_OFFSET),
      [tsync] "i" ((uint8_t)DEJITTER_SYNC)
    : "r30", "r31", "r16", "r17");

maujabur, 2016-09-14:
probably the problem is here

//" adc r31, zero_reg \n\t" //

i'll test it later this week

abdul-rehman-2050, 2017-04-21:
Did any body get this worked on mega?

snobu, 2017-04-21:
Nope. I just i2c'd together a Mega and a Uno and used the Uno for VGA.

abdul-rehman-2050, 2017-04-21:
is this project dead? 
Not seemed any response from author so long

snobu, 2017-04-21:
I think it's awesome and short of magic, not dead. Maintaining things is a best effort.. Effort.

snobu, 2017-10-26:
@smaffer @maujabur Just throwing this out there... what about the new SAMD based chips like the M0 - Arduino Zero/Adafruit Feather M0? Is there a chance we could get 640x480? That's my 4k standard for Arduino VGA. How much different is the assembly code from the AVR? 100%? :)

maujabur, 2017-10-26:
@snobu kt really is 100% different on assembly, I'll look for some library and let you know if I can find something.

KaMeCache, 2018-03-05:
@maujabur You have written that it works on your Mega2650.
I got the H and V SYNC working, with
#define HSYNCPIN 9
#define VSYNCPIN 11

But i still get only black on the screen,
i tried all Output Pins connected to bit 6 and bit7 on the board.

Outcommenting of jitter did non help/work for me.
(can't compile)

What did you do, exactly?

smaffer, 2018-08-21:
If someone has some working code, may create a branch and upload the code here,  code will not be lost :)

smaffer, 2019-04-13:
I'M BACK! :D
i've found all the problems with atmega2560, i'm now working on it :)
These are the required changes:

    #define HSYNCPIN 9
    #define VSYNCPIN 11
    #define COLORPIN0 30
    #define COLORPIN1 31

for interrupt dejitter, atmega2560 seems to use a 22bit program counter, so the code must be changed like this:

      "     call TL               \n\t" //
      "TL:                        \n\t" //
      "     pop r17               \n\t" //
      "     pop r31               \n\t" //
      "     pop r30               \n\t" //

Timer0 setup must be changed to

     //disable TIMER0 interrupt
     TIMSK0=0;
     TCCR0A=0;
     TCCR0B=(1 << CS00); //enable 16MHz counter (used to fix the HSYNC interrupt jitter)
     OCR0A=0;
     OCR0B=0;
     TCNT0=0;

Also PORTD must be changed to PORTC

I'm planning to release a new version that support both ATMEGA328 and ATMEGA2560. With ATMEGA2560 the vertical resolution can be increased to 90px (4:3 aspect ratio) or more. Coming soon!

On ATMEGA2560 is possible to achive 120x60px with 256 colors (RRRGGGBB)... is possible but need big changes to the library and tools. 120x90px with 16 colors (RGBH-RGBH) is also possible... I will think if have sense to write another library with these variations or write them on VGAX.

Thank you!

smaffer, 2019-04-14:
Hello! 
I've released a new version with ATMega2560 support. Resolution can be increased to 120x90px with squared pixels or 120x240px with rectangular pixels.
Have fun!!

smaffer, 2019-04-14:
[https://www.youtube.com/watch?v=Qs3uyhzUcfA](url) 120x240px example :)

closed 2019-04-14 by smaffer

vrxfile, 2019-04-14:
Hi! Thank you very much!

⁣Best regards,
Rostislav Varzar

п>Hello!