smaffer/vgax #7: Higher Resolution, Monochrome?

himijendrix24, 2016-10-22:
Hello, 
at the moment I'm playing around with your library and would like to use it for generating geometrical pixel art with the TVOut utils. Would it actually be possible to get a higher resolution (800x600) and therefore leave the colour? 
Greetings,
Himi

NateDogg1232, 2016-11-27:
I believe this project is dead. It's last commit was in 2015.

silviustro, 2016-11-27:
Here's a talk I had with the creator of the project explaining the trouble with increasing the resolution.

> the problem with the resolution is this: without overclocking, running at 16mhz, is very difficult to increase the number of pixels generated in one line.. so VGAX_WIDTH can’t be easily increased.
If you use an atmega chip with more RAM, is possible to increase the VGAX_HEIGHT but i believe that you need to modify some part of the code. I’ve received some email from others programmers that have tried to port VGAX to ATMEGA2560 but some interrupts and some pins need to be changed. 

> At 16Mhz, using at least 2 instructions-per-pixels (lds+out), is impossible to generate more pixels.. The standard VGA signal runs at 25Mhz and, without overcloking, the MCU is too slow to send out more than 160 or 180 pixels. 120 is a safer resolution that i have tested on some monitors.

smaffer, 2016-11-27:
Hi,
that's right silviustro. 800x600 is not possible at 16mhz, without any
additional hardware.
himijendrix24 you can try with gameduino or some boards that have a
dedicated CPU. For what i remember Gameduino2 has hardware 3d acceleration
too.

smaffer, 2018-08-21:
@silviustro take a look to my new library ESPVGAX, for ESP8266, it has 512x480px 1bpp framebuffer :)

closed 2018-08-21 by smaffer