smaffer/vgax #10: Squeeze the bitmaps required format (idea), one byte for two pixels and cache method

codebeat-nl, 2017-08-26:
Very nice, thanks for sharing, seems to compile also on the Nano which made it possible to use it as a little VGA driver (via for example I2C interface) for an another Arduino. 

Design comment: The bitmaps uses four bytes for each pixel. Because you have only four colors, why you didn''t designed it to use color numbers that reffers to a palette like GIF's (or other <=256 color bitmaps) do. So you need just one byte for each pixel and saves you 3x times the required flash memory, for example: 

0 : Black 
1 : Color #1 
2: Color #2 
3: Color #3 
4: Color #4 

You can also squeeze 2 pixels in one byte. The first four bits specify the color of the first pixel and the last bits specify the color of the second pixel. This halfs the required space, for example (bin format): 
1000 : This selects color #1 for pixel one 
0001 : This selects color #4 for pixel two 

So in the array, it looks something like this: B10000001, which is just one byte = 129 or 0x81 hex. Hex notation makes the code a bit smalller ofcourse. 

Because this all saves space, maybe there is room to apply chaching to avoid the flicker. The cache can be build the same like the bitmaps, so it requires less memory and the timer draws the cache continuously in background. So when you manipulate the output by drawing a pixel, you draw in cache instead of draw it directly. You also can 'move' portions directly in memory without any translation required which makes it as fast as possible. 

Idea?﻿

fantostisch, 2018-05-28:
The bitmaps use 2 bits for each pixel, not 4 bytes.

smaffer, 2018-08-21:
@fantostisch is right, VGAX already uses 2bits for pixel, packing 4 pixels in 1 byte. I think that a palette approach is not a better solution, due to the limitation of 16Mhz speed. The pixel output loop uses two instructions:
```
ld r16, Z+
out %[port], r16
```
with a total of 3 CPU cycles (see http://ww1.microchip.com/downloads/en/devicedoc/atmel-0856-avr-instruction-set-manual.pdf). A lookup to a palette table should require too time. Keep in mind that more time you spent generating pixels colors, less pixels you can output on each row!

closed 2018-08-21 by smaffer